Arrow-air/project-quiver #29: 04/07 - Flight Test Setup

alperenag, 2025-04-03:
- Creating a flight test setup similar to one we have in Texas

- "arrow-ge" discord account

- Order 1 more battery

Julius-eng, 2025-04-07:
- Battery is ordered
- I also ordered a charger

Julius-eng, 2025-04-08:
To get to the test site, I have to walk about 50-100m across terrain. To create a setup similar to the one in texas, I decided on the following:
- Ordered a handcart with large wheels. I will build a second level on top so that it is easy to work on. There is space on the lower level to transport the drone at the same time.
- Ordered a monitor with gas spring mount
- Ordered a mini pc
- Ordered a holder for a dslr camera that can be adjusted quickly
- I still have a DSLR camera, which is not used often and can be used permanently for the setup

It will take a few days until all the things arrive and the table setup is ready. I'll try to have everything ready for Monday 14.04.

closed 2026-01-29 by thomasgarrison