Arrow-air/project-quiver #30: 04/21 - Flight Test Log Analyzer

alperenag, 2025-04-03:


alexdada555, 2025-04-10:
#Update 

System developed in python as a web application extracting and producing plots of the following parameters

## Estimation Status (EKF3/Normalised Innovations EK3 Lane1)

- XKF4[0].SV
- XKF4[0].SP
- XKF4[0].SH
- XKF4[0].SM
- XKF4[0].SVT

## GPS Status (Sensors/GPS/GPS Accuracy)

- GPA[0].HAcc
- GPA[0].SAcc
- GPA[0].VAcc

## Vibration Metrics (Sensors/Accelerometer/Vibration)

- VIBE[0].VibeX
- VIBE[0].VibeY
- VIBE[0].VibeZ

## Clipping( Sensors/Accelerometer/Clipping)

- VIBE[0].Clip
- VIBE[1].Clip

## Rate Controller Performance

- RATE.R and RATE.RDes
- RATE.P and RATE.PDes
- RATE.Y and RATE.YDes

## Attitude Controller Performance

- ATT.Roll and ATT.DesRoll
- ATT.Pitch and ATT.DesPitch
- ATT.Yaw and ATT.DesYaw

## PWM Outputs

- RCOU.C1
- RCOU.C2
- RCOU.C3
- RCOU.C4

## RC Inputs

- RCIN.C1
- RCIN.C2
- RCIN.C3
- RCIN.C4

## Battery related parameters

- BAT[0].Volt
- BAT[0].Curr
- BAT[0].Temp 

---
We can add more later but these should be sufficient for now. 

These are being produced as individual graph images and displayed on a web page
Core functionality is ready

# Currently 

Exploring how best to deploy it to a remote server for accessibility and scalability to multi  user data capture and storage

More functionality can be added in late

alexdada555, 2025-04-15:
# Update

Test Implementation Repository Created and functional  Local deployment produced

[Flight Log Analyser Repository](https://github.com/alexdada555/Flight-Log-Analyser)

alexdada555, 2025-04-29:
Application Updates in coming with `.Log` Files parsing functionality using this library https://pypi.org/project/pymavlog/

closed 2026-01-29 by thomasgarrison